lnis-uofu/LSOracle #14: build problems in kahypar

sheriefreda, 2019-12-03:
I am having build problems, errors arising in building kayhypar library. When I download kayhypar separately from its own repo, it compiles fine. The error is coming from the folder that is included with LSOracle. I am using macOSX 10.14.4

/Users/sheriefreda/Tools/LSOracle/lib/kahypar/kahypar/partition/refinement/flow/most_balanced_minimum_cut.h:372:10: error: no member named 'random_shuffle' in namespace 'std'
    std::random_shuffle(start_nodes.begin(), start_nodes.end());
    ~~~~~^

srtemp, 2019-12-03:
Hi Sherief,
Thanks for bringing this up. It looks like we're using an older version of KaHyPar which is using the deprecated random_shuffle function, while the version in their master repo is using std::shuffle. I'm not sure why it's building on my end, it must be a slightly older standard library than you're using.  I'll update the code on our end push a fix this afternoon, assuming no big problems with upgrading kahypar. I'll update this and close the issue when that's done.
Thanks again,
Scott

sheriefreda, 2019-12-06:
Hi, I see you closed the issue, but there are no commits on the master branch. Please let me know the status. Thanks a lot for the support.

srtemp, 2019-12-06:
Hi, 
This still appears open on my end, and we're still working on it.  We have a patch in a feature branch, but we are doing some testing before merging to master.  Sorry for the delay, we have some customization to KaHyPar, so it is not quite as simple as I hoped.
Scott

srtemp, 2019-12-06:
We've merged in the change; please let us know if that resolves your problem.

Scott

sheriefreda, 2019-12-08:
thanks a lot;  the error with kahypar is now gone. I am getting this error now (8 of them)

In file included from /Users/sheriefreda/Tools/LSOracle/core/lsoracle.cpp:56:
/Users/sheriefreda/Tools/LSOracle/core/commands/output/get_all_partitions.hpp:83:62: error: 'path' is unavailable: introduced in macOS
      10.15
              std::string toplevel_module = std::filesystem::path(ntk._storage->net_name).filename();
                                                             ^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/filesystem:739:24: note: 
      'path' has been explicitly marked unavailable here
class _LIBCPP_TYPE_VIS path {
                       ^
In file included from /Users/sheriefreda/Tools/LSOracle/core/lsoracle.cpp:56:
/Users/sheriefreda/Tools/LSOracle/core/commands/output/get_all_partitions.hpp:83:45: error: 'path' is unavailable: introduced in macOS
      10.15
              std::string toplevel_module = std::filesystem::path(ntk._storage->net_name).filename();
                                            ^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/filesystem:773:3: note: 'path'
      has been explicitly marked unavailable here
  path(const _Source& __src, format = format::auto_format) {
  ^
In file included from /Users/sheriefreda/Tools/LSOracle/core/lsoracle.cpp:56:
/Users/sheriefreda/Tools/LSOracle/core/commands/output/get_all_partitions.hpp:83:45: error: '~path' is unavailable: introduced in macOS
      10.15
              std::string toplevel_module = std::filesystem::path(ntk._storage->net_name).filename();
                                            ^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/filesystem:791:3: note: 
      '~path' has been explicitly marked unavailable here
  ~path() = default;
  ^
In file included from /Users/sheriefreda/Tools/LSOracle/core/lsoracle.cpp:56:
/Users/sheriefreda/Tools/LSOracle/core/commands/output/get_all_partitions.hpp:83:91: error: 'filename' is unavailable: introduced in
      macOS 10.15
              std::string toplevel_module = std::filesystem::path(ntk._storage->net_name).filename();
                                                                                          ^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/filesystem:1067:34: note: 
      'filename' has been explicitly marked unavailable here
  _LIBCPP_INLINE_VISIBILITY path filename() const {
                                 ^
In file included from /Users/sheriefreda/Tools/LSOracle/core/lsoracle.cpp:56:
/Users/sheriefreda/Tools/LSOracle/core/commands/output/get_all_partitions.hpp:83:91: error: '~path' is unavailable: introduced in macOS
      10.15
              std::string toplevel_module = std::filesystem::path(ntk._storage->net_name).filename();
                                                                                          ^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/filesystem:791:3: note: 
      '~path' has been explicitly marked unavailable here
  ~path() = default;
  ^
In file included from /Users/sheriefreda/Tools/LSOracle/core/lsoracle.cpp:56:
/Users/sheriefreda/Tools/LSOracle/core/commands/output/get_all_partitions.hpp:83:27: error: 'operator basic_string' is unavailable:
      introduced in macOS 10.15
              std::string toplevel_module = std::filesystem::path(ntk._storage->net_name).filename();
                          ^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/filesystem:985:29: note: 
      'operator basic_string' has been explicitly marked unavailable here
  _LIBCPP_INLINE_VISIBILITY operator string_type() const { return __pn_; }
                            ^
In file included from /Users/sheriefreda/Tools/LSOracle/core/lsoracle.cpp:56:
/Users/sheriefreda/Tools/LSOracle/core/commands/output/get_all_partitions.hpp:97:59: error: 'path' is unavailable: introduced in macOS
      10.15
                std::string modulename = std::filesystem::path( ntk._storage->net_name + "_" + std::to_string(partition) ).filename();
                                                          ^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/filesystem:739:24: note: 
      'path' has been explicitly marked unavailable here
class _LIBCPP_TYPE_VIS path {
                       ^
In file included from /Users/sheriefreda/Tools/LSOracle/core/lsoracle.cpp:56:
/Users/sheriefreda/Tools/LSOracle/core/commands/output/get_all_partitions.hpp:97:42: error: 'path' is unavailable: introduced in macOS
      10.15
                std::string modulename = std::filesystem::path( ntk._storage->net_name + "_" + std::to_string(partition) ).filename();
                                         ^
[... 13 lines cut ...]
In file included from /Users/sheriefreda/Tools/LSOracle/core/lsoracle.cpp:56:
/Users/sheriefreda/Tools/LSOracle/core/commands/output/get_all_partitions.hpp:97:124: error: 'filename' is unavailable: introduced in
      macOS 10.15
                std::string modulename = std::filesystem::path( ntk._storage->net_name + "_" + std::to_string(partition) ).filename();
[... cut ...]
/Users/sheriefreda/Tools/LSOracle/core/commands/output/get_all_partitions.hpp:97:29: error: 'operator basic_string' is unavailable:
      introduced in macOS 10.15
                std::string modulename = std::filesystem::path( ntk._storage->net_name + "_" + std::to_string(partition) ).filename();
                            ^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/filesystem:985:29: note: 
      'operator basic_string' has been explicitly marked unavailable here

srtemp, 2019-12-10:
I'm building a clean 10.14 VM to try to replicate this now, but I suspect that this will be resolved if you build with gcc.  I'll get back to you once I reproduce the errors above, but if you prefer not to wait, you can try this.

brew install gcc
brew install --cc=gcc-9 boost

cmake .. -DCMAKE_CXX_COMPILER=/usr/local/bin/g++-9 -DCMAKE_CC_COMPILER=/usr/local/bin/gcc-9 -DCMAKE_BUILD_TYPE=Release
make

srtemp, 2019-12-11:
I haven't been able to duplicate your errors exactly, but on a fresh 10.14.4 install using apple clang I got a different, presumably related error from std::filesystem.  Using gcc resolved the issue; however, the current version of boost in homebrew, 1.71.0, seems to ignore the cc=gcc-9 flag, causing problems.  I'm still looking into that; in the interim, I recommend using boost@1.60, which worked smoothly.

I'll update the documentation to have detailed mac instructions; let me know if you run into any more problems.

Scott

sheriefreda, 2019-12-18:
Hi Scott, I was on travel and came back. I tried the compile instructions you mentioned and did not work. If you have success with the Mac build, please let know.

yashton, 2021-06-14:
PR #39 tests build with Apple clang, closing stale issue.

closed 2021-06-14 by yashton