lnis-uofu/LSOracle #17: building issue of filesystem header

Ace-Ma, 2020-01-21:
Hi. I'm trying to rebuild the newest version of LSOracle, and stuck in an error from <filesystem>. The log shows below:
`In file included from /home/jingxiao/Tools/LSOracle/core/lsoracle.cpp:56:0:
/home/jingxiao/Tools/LSOracle/core/commands/output/get_all_partitions.hpp:17:10: fatal error: filesystem: No such file or directory
 #include <filesystem>
          ^~~~~~~~~~~~
compilation terminated.
core/CMakeFiles/lsoracle.dir/build.make:62: recipe for target 'core/CMakeFiles/lsoracle.dir/lsoracle.cpp.o' failed
make[2]: *** [core/CMakeFiles/lsoracle.dir/lsoracle.cpp.o] Error 1
CMakeFiles/Makefile2:1217: recipe for target 'core/CMakeFiles/lsoracle.dir/all' failed
make[1]: *** [core/CMakeFiles/lsoracle.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2`

Could you please have a look? Thank you.

srtemp, 2020-01-21:
Hello; thanks for bringing this up.
Which compiler (and which version) are you building with?
Filesystem is a c++ 17 feature, and should work fine with gcc 8 or 9.  On gcc7 it should work if you turn on the experimental c++ 17 features with the -std=c++1z flag.
If you have an installation of gcc8+ which isn't the default gcc/g++, you can pass it to cmake with -DCMAKE_CXX_COMPILER=/path/to/g++ and -DCMAKE_CC_COMPILER=/path/to/gcc

I hope that helps; let me know if you have any other problems.

Scott

yashton, 2021-06-14:
PR #39 updates the cmake file to check for compatibility and add the compiler flag as needed.

closed 2021-06-14 by yashton