lnis-uofu/LSOracle #18: CMAKE dependancy requirement

aolofsson, 2020-01-22:
README refers to cmake 3.9 being required, but it looks like kahypar requires 3.12

aolofsson, 2020-01-22:
The following script was needed in Ubuntu 18.04. resulting executable in build/core/lsoracle seems to work.

```bash
Install Latest Cmake
sudo apt-get install libssl-dev
wget https://github.com/Kitware/CMake/releases/download/v3.16.3/cmake-3.16.3.tar.gz
tar zvf cmake-3.16.3.tar.gz
cd cmake-3.16.3
./bootstrap
make
sudo make install

#Adding g++ 8.x
sudo add-apt-repository ppa:ubuntu-toolchain-r/test
sudo apt install g++-8
sudo update-alternatives --install /usr/bin/gcc gcc /usr/bin/gcc-8 60 --slave /usr/bin/g++ g++ /usr/bin/g++-8

#Build LS Oracle
sudo apt-get install libreadline-dev
git clone --recursive git@github.com:LNIS-Projects/LSOracle.git --branch demo
mkdir build
cd build
cmake .. -DCMAKE_BUILD_TYPE=RELEASE
make

```

srtemp, 2020-01-23:
Hi Andreas,
We've pulled demo into master, minus a few things to keep mac compatibility, and updated the readme, including dependency information.

Thanks for bringing this up; it was definitely stale.

Scott

closed 2020-01-23 by srtemp