lnis-uofu/LSOracle #20: Instructions for using as a Yosys plugin?

mithro, 2020-10-25:
The https://woset-workshop.github.io/WOSET2020.html paper claims LSOracle can be used as a Yosys plugin but the README doesn't mention this (or Yosys at all)? Also couldn't find any reference to Yosys in the documentation at https://lsoracle.readthedocs.io/en/master/?badge=master ?

Ravenslofty, 2021-03-11:
There's a [branch](https://github.com/lnis-uofu/LSOracle/tree/yosys-plugin) with those instructions. It's a bit hard to find though.

yashton, 2021-03-11:
The branch Ravenslofty references is no longer the source for the plugin, it has been moved to https://github.com/lnis-uofu/LSOracle-Plugin
We can add more instructions to the readme and to the documentation. If there is anything in particular you need, let me know.

closed 2021-06-01 by yashton