lnis-uofu/LSOracle #37: LSOracle crashes with message

ratul619, 2021-06-09:
Hello , 

I am trying to load below example file and run oracle on it. But the shell crashes. With the message "Could not load context file at: /usr/local/share/lsoracle/test.ini" 

lsoracle> read lib/mockturtle/test/benchmarks/c3540.aig
AIG network stored
lsoracle> oracle
Could not load context file at: /usr/local/share/lsoracle/test.ini

Can you please help regarding the same ?

yashton, 2021-06-09:
By default, the configuration file is looked for at that location. This was updated several months ago from looking for a file in a relative path from the build directory ('../../core/test.ini'), which was non-portable. You can copy the file in core/test.ini to that location (which is done during make installation), or you can specify the config file using the --config or -f flags on the oracle command.
I will update the documentation to reflect this.

ratul619, 2021-06-09:
Thanks Ashton . The program started running after using below command.

lsoracle> oracle --config core/test.ini

yashton, 2021-07-13:
In PR #57 I setup the config file to be automatically generated, which should alleviate any confusion about specifying this file.

closed 2021-07-13 by yashton