lnis-uofu/LSOracle #41: Read bllif error

ratul619, 2021-06-13:
Hi , 

My blif file contains the below lines : 

.model TopLevel432b
.inputs E[0] E[1] E[2] E[3] E[4] E[5] E[6] E[7] E[8] A[0] A[1] A[2] A[3] A[4] A[5] A[6] A[7] A[8] B[0] B[1] B[2] B[3] B[4] B[5] B[6] B[7] B[8] C[0] C[1] C[2] C[3] C[4] C[5] C[6] C[7] C[8]
.outputs PA PB PC Chan[0] Chan[1] Chan[2] Chan[3]
.names $false
.names $true
1
.names $undef
.subckt PriorityA A[0]=A[0] A[1]=A[1] A[2]=A[2] A[3]=A[3] A[4]=A[4] A[5]=A[5] A[6]=A[6] A[7]=A[7] A[8]=A[8] E[0]=E[0] E[1]=E[1] E[2]=E[2] E[3]=E[3] E[4]=E[4] E[5]=E[5] E[6]=E[6] E[7]=E[7] E[8]=E[8] PA=PA X1[0]=X1[0] X1[1]=X1[1] X1[2]=X1[2] X1[3]=X1[3] X1[4]=X1[4] X1[5]=X1[5] X1[6]=X1[6] X1[7]=X1[7] X1[8]=X1[8]
.subckt PriorityB B[0]=B[0] B[1]=B[1] B[2]=B[2] B[3]=B[3] B[4]=B[4] B[5]=B[5] B[6]=B[6] B[7]=B[7] B[8]=B[8] E[0]=E[0] E[1]=E[1] E[2]=E[2] E[3]=E[3] E[4]=E[4] E[5]=E[5] E[6]=E[6] E[7]=E[7] E[8]=E[8] PB=PB X1[0]=X1[0] X1[1]=X1[1] X1[2]=X1[2] X1[3]=X1[3] X1[4]=X1[4] X1[5]=X1[5] X1[6]=X1[6] X1[7]=X1[7] X1[8]=X1[8] X2[0]=X2[0] X2[1]=X2[1] X2[2]=X2[2] X2[3]=X2[3] X2[4]=X2[4] X2[5]=X2[5] X2[6]=X2[6] X2[7]=X2[7] X2[8]=X2[8]
.subckt PriorityC C[0]=C[0] C[1]=C[1] C[2]=C[2] C[3]=C[3] C[4]=C[4] C[5]=C[5] C[6]=C[6] C[7]=C[7] C[8]=C[8] E[0]=E[0] E[1]=E[1] E[2]=E[2] E[3]=E[3] E[4]=E[4] E[5]=E[5] E[6]=E[6] E[7]=E[7] E[8]=E[8] PC=PC X1[0]=X1[0] X1[1]=X1[1] X1[2]=X1[2] X1[3]=X1[3] X1[4]=X1[4] X1[5]=X1[5] X1[6]=X1[6] X1[7]=X1[7] X1[8]=X1[8] X2[0]=X2[0] X2[1]=X2[1] X2[2]=X2[2] X2[3]=X2[3] X2[4]=X2[4] X2[5]=X2[5] X2[6]=X2[6] X2[7]=X2[7] X2[8]=X2[8]
.subckt EncodeChan A[0]=A[0] A[1]=A[1] A[2]=A[2] A[3]=A[3] A[4]=A[4] A[5]=A[5] A[6]=A[6] A[7]=A[7] A[8]=A[8] B[0]=B[0] B[1]=B[1] B[2]=B[2] B[3]=B[3] B[4]=B[4] B[5]=B[5] B[6]=B[6] B[7]=B[7] B[8]=B[8] C[0]=C[0] C[1]=C[1] C[2]=C[2] C[3]=C[3] C[4]=C[4] C[5]=C[5] C[6]=C[6] C[7]=C[7] C[8]=C[8] E[0]=E[0] E[1]=E[1] E[2]=E[2] E[3]=E[3] E[4]=E[4] E[5]=E[5] E[6]=E[6] E[7]=E[7] E[8]=E[8] I[0]=I[0] I[1]=I[1] I[2]=I[2] I[3]=I[3] I[4]=I[4] I[5]=I[5] I[6]=I[6] I[7]=I[7] I[8]=I[8] PA=PA PB=PB PC=PC
.subckt DecodeChan Chan[0]=Chan[0] Chan[1]=Chan[1] Chan[2]=Chan[2] Chan[3]=Chan[3] I[0]=I[0] I[1]=I[1] I[2]=I[2] I[3]=I[3] I[4]=I[4] I[5]=I[5] I[6]=I[6] I[7]=I[7] I[8]=I[8].end

But reading in lsoracle is giving errors as below: Is the .subckt definition for blif supported ? 

[e] cannot parse line `.subckt TopLevel432b A[0]=in102 A[1]=in89 A[2]=in76 A[3]=in63 A[4]=in50 A[5]=in37 A[6]=in24 A[7]=in11 A[8]=in1 B[0]=in112 B[1]=in99 B[2]=in86 B[3]=in73 B[4]=in60 B[5]=in47 B[6]=in34 B[7]=in21 B[8]=in8 C[0]=in115 C[1]=in105 C[2]=in92 C[3]=in79 C[4]=in66 C[5]=in53 C[6]=in40 C[7]=in27 C[8]=in14 Chan[0]=Chan[0] Chan[1]=Chan[1] Chan[2]=Chan[2] Chan[3]=Chan[3] E[0]=in108 E[1]=in95 E[2]=in82 E[3]=in69 E[4]=in56 E[5]=in43 E[6]=in30 E[7]=in17 E[8]=in4 PA=PA PB=PB PC=PC`
[e] cannot parse line `.subckt PriorityA A[0]=A[0] A[1]=A[1] A[2]=A[2] A[3]=A[3] A[4]=A[4] A[5]=A[5] A[6]=A[6] A[7]=A[7] A[8]=A[8] E[0]=E[0] E[1]=E[1] E[2]=E[2] E[3]=E[3] E[4]=E[4] E[5]=E[5] E[6]=E[6] E[7]=E[7] E[8]=E[8] PA=PA X1[0]=X1[0] X1[1]=X1[1] X1[2]=X1[2] X1[3]=X1[3] X1[4]=X1[4] X1[5]=X1[5] X1[6]=X1[6] X1[7]=X1[7] X1[8]=X1[8]`
[e] cannot parse line `.subckt PriorityB B[0]=B[0] B[1]=B[1] B[2]=B[2] B[3]=B[3] B[4]=B[4] B[5]=B[5] B[6]=B[6] B[7]=B[7] B[8]=B[8] E[0]=E[0] E[1]=E[1] E[2]=E[2] E[3]=E[3] E[4]=E[4] E[5]=E[5] E[6]=E[6] E[7]=E[7] E[8]=E[8] PB=PB X1[0]=X1[0] X1[1]=X1[1] X1[2]=X1[2] X1[3]=X1[3] X1[4]=X1[4] X1[5]=X1[5] X1[6]=X1[6] X1[7]=X1[7] X1[8]=X1[8] X2[0]=X2[0] X2[1]=X2[1] X2[2]=X2[2] X2[3]=X2[3] X2[4]=X2[4] X2[5]=X2[5] X2[6]=X2[6] X2[7]=X2[7] X2[8]=X2[8]`
[e] cannot parse line `.subckt PriorityC C[0]=C[0] C[1]=C[1] C[2]=C[2] C[3]=C[3] C[4]=C[4] C[5]=C[5] C[6]=C[6] C[7]=C[7] C[8]=C[8] E[0]=E[0] E[1]=E[1] E[2]=E[2] E[3]=E[3] E[4]=E[4] E[5]=E[5] E[6]=E[6] E[7]=E[7] E[8]=E[8] PC=PC X1[0]=X1[0] X1[1]=X1[1] X1[2]=X1[2] X1[3]=X1[3] X1[4]=X1[4] X1[5]=X1[5] X1[6]=X1[6] X1[7]=X1[7] X1[8]=X1[8] X2[0]=X2[0] X2[1]=X2[1] X2[2]=X2[2] X2[3]=X2[3] X2[4]=X2[4] X2[5]=X2[5] X2[6]=X2[6] X2[7]=X2[7] X2[8]=X2[8]`
[e] cannot parse line `.subckt EncodeChan A[0]=A[0] A[1]=A[1] A[2]=A[2] A[3]=A[3] A[4]=A[4] A[5]=A[5] A[6]=A[6] A[7]=A[7] A[8]=A[8] B[0]=B[0] B[1]=B[1] B[2]=B[2] B[3]=B[3] B[4]=B[4] B[5]=B[5] B[6]=B[6] B[7]=B[7] B[8]=B[8] C[0]=C[0] C[1]=C[1] C[2]=C[2] C[3]=C[3] C[4]=C[4] C[5]=C[5] C[6]=C[6] C[7]=C[7] C[8]=C[8] E[0]=E[0] E[1]=E[1] E[2]=E[2] E[3]=E[3] E[4]=E[4] E[5]=E[5] E[6]=E[6] E[7]=E[7] E[8]=E[8] I[0]=I[0] I[1]=I[1] I[2]=I[2] I[3]=I[3] I[4]=I[4] I[5]=I[5] I[6]=I[6] I[7]=I[7] I[8]=I[8] PA=PA PB=PB PC=PC`
[e] cannot parse line `.subckt DecodeChan Chan[0]=Chan[0] Chan[1]=Chan[1] Chan[2]=Chan[2] Chan[3]=Chan[3] I[0]=I[0] I[1]=I[1] I[2]=I[2] I[3]=I[3] I[4]=I[4] I[5]=I[5] I[6]=I[6] I[7]=I[7] I[8]=I[8]`
parsing failed

Thanks !

yashton, 2021-06-14:
I got the same error when i tried reading your file into LSOracle. I also got a similar error reading it in yosys:
```
yosys> read_blif test.blif
1. Executing BLIF frontend.
ERROR: Syntax error in line 13!
```

ratul619, 2021-06-19:
Yeah. Strange thing is the blif is generated from Yosys itself.

yashton, 2021-06-21:
Weird. Could you post your yosys script and input file?

srtemp, 2022-07-07:
We use [lorina](https://github.com/hriener/lorina) for native parsing, although you could also use the Yosys plugin.  Lorina doesn't support .subckt.  I was able to get Yosys to read your file by moving .end to a new line.  I'd recommend using the Yosys plugin to run LSOracle on this file.

closed 2022-07-07 by srtemp