lnis-uofu/LSOracle #47: Regarding output ports mapping to BLIF/verilog

ratul619, 2021-06-19:
Hi , 

In the  dot file generated , the outputs are named as po*. But in the verilog/blif files as input , the names are of course different for both inputs and outputs. 

How to know the mapping of input/output for input blif/verilog to the dot file ? 

108 -> po3 [style=solid]
111 -> po4 [style=solid]
120 -> po5 [style=dashed]
122 -> po6 [style=solid]
127 -> po7 [style=solid]
{rank = same; po0; po1; po2; po3; po4; po5; po6; po7; }

Thanks !

ratul619, 2021-06-24:
Any feedback on this ?

yashton, 2021-06-24:
Unfortunately the labels in the graph are not tied to the names coming in, but are pretty much arbitrary. This graphing is handled by mockturtle.

NormandAlexandre, 2022-07-01:
Hey,
The input names are now preserved in the dot file; the PO names are numbered in the order they are given in the verilog.

closed 2022-07-01 by NormandAlexandre