lnis-uofu/LSOracle #74: Confusion about command "techmap"

foggy-slt, 2022-03-09:
Hi
I'm having a problem with the LSOracle.

This is my step：
lsoracle> read_aig 0.aig
lsoracle> oracle
lsoracle> write_blif -m 0.blif

But I am using the "cec" of the tool "ABC", and the verificationis not equivalent.

abc 01> cec 0.aig 0.blif
Networks are NOT EQUIVALENT.  Time =     0.00 sec
Verification failed for at least 8 outputs:  output_47 output_50 output_53 ...
Output output_47: Value in Network1 = 0. Value in Network2 = 1.
Input pattern:  input_45=0 input_46=0 input_44=0

I have tried other cases, and this situation will also occur.
Upload a case below.
Thanks!
[0.zip](https://github.com/lnis-uofu/LSOracle/files/8212796/0.zip)

yashton, 2022-04-20:
could you post your circuit?

foggy-slt, 2022-04-21:
I uploaded a zip package named 0.zip in the first comment, you can unzip it and use it, thanks.

NormandAlexandre, 2022-06-16:
Hi,
We have updated the oracle command and it’s working now.
My steps : 
lsoracle> read_aig 0.aig
lsoracle> oracle
lsoracle> write_blif -g 0.blif 

Note that the oracle command writes a xmg by default.
The two networks are equivalent.

closed 2022-06-16 by NormandAlexandre