lnis-uofu/LSOracle #80: make error

ratul619, 2022-04-18:
Hi , 

Getting below error while make command : 

[ 75%] Built target kahypar
[ 89%] Built target unit_tests
[ 89%] Building CXX object core/CMakeFiles/lsoracle.dir/lsoracle.cpp.o
In file included from /home/ratul619/LSoracle/LSOracle/core/lsoracle.cpp:145:
/home/ratul619/LSoracle/LSOracle/core/commands/optimization/optimize.hpp: In member function ‘void alice::optimize_command::synth(std::string)’:
/home/ratul619/LSoracle/LSOracle/core/commands/optimization/optimize.hpp:76:27: error: ‘optimize_resynthesis’ is not a member of ‘oracle’
   76 |      ntk_result = oracle::optimize_resynthesis<mockturtle::aig_network>(partitions_jr, abc_exec);
      |                           ^~~~~~~~~~~~~~~~~~~~
/home/ratul619/LSoracle/LSOracle/core/commands/optimization/optimize.hpp:76:71: error: expected primary-expression before ‘>’ token
   76 |      ntk_result = oracle::optimize_resynthesis<mockturtle::aig_network>(partitions_jr, abc_exec);
      |                                                                       ^
/home/ratul619/LSoracle/LSOracle/core/commands/optimization/optimize.hpp:78:14: error: ‘optimization_strategy’ is not a member of ‘oracle’; did you mean ‘optimization_test’?
   78 |      oracle::optimization_strategy strategy;
      |              ^~~~~~~~~~~~~~~~~~~~~
      |              optimization_test
/home/ratul619/LSoracle/LSOracle/core/commands/optimization/optimize.hpp:80:3: error: ‘strategy’ was not declared in this scope
   80 |   strategy = oracle::optimization_strategy::depth;
      |   ^~~~~~~~
/home/ratul619/LSoracle/LSOracle/core/commands/optimization/optimize.hpp:80:22: error: ‘oracle::optimization_strategy’ has not been declared
   80 |   strategy = oracle::optimization_strategy::depth;
      |                      ^~~~~~~~~~~~~~~~~~~~~
/home/ratul619/LSoracle/LSOracle/core/commands/optimization/optimize.hpp:82:3: error: ‘strategy’ was not declared in this scope
   82 |   strategy = oracle::optimization_strategy::size;
      |   ^~~~~~~~
/home/ratul619/LSoracle/LSOracle/core/commands/optimization/optimize.hpp:82:22: error: ‘oracle::optimization_strategy’ has not been declared
   82 |   strategy = oracle::optimization_strategy::size;
      |                      ^~~~~~~~~~~~~~~~~~~~~
/home/ratul619/LSoracle/LSOracle/core/commands/optimization/optimize.hpp:84:3: error: ‘strategy’ was not declared in this scope
   84 |   strategy = oracle::optimization_strategy::balanced;
      |   ^~~~~~~~
/home/ratul619/LSoracle/LSOracle/core/commands/optimization/optimize.hpp:84:22: error: ‘oracle::optimization_strategy’ has not been declared
   84 |   strategy = oracle::optimization_strategy::balanced;
      |                      ^~~~~~~~~~~~~~~~~~~~~
/home/ratul619/LSoracle/LSOracle/core/commands/optimization/optimize.hpp:87:34: error: ‘optimize_basic’ is not a member of ‘oracle’
   87 |             ntk_result = oracle::optimize_basic<network>(partitions_jr, abc_exec, strategy);
      |                                  ^~~~~~~~~~~~~~
/home/ratul619/LSoracle/LSOracle/core/commands/optimization/optimize.hpp:87:56: error: expected primary-expression before ‘>’ token
   87 |             ntk_result = oracle::optimize_basic<network>(partitions_jr, abc_exec, strategy);
      |                                                        ^
/home/ratul619/LSoracle/LSOracle/core/commands/optimization/optimize.hpp:87:83: error: ‘strategy’ was not declared in this scope
   87 |             ntk_result = oracle::optimize_basic<network>(partitions_jr, abc_exec, strategy);
      |                                                                                   ^~~~~~~~

In file included from /home/ratul619/LSoracle/LSOracle/core/lsoracle.cpp:63:
/home/ratul619/LSoracle/LSOracle/core/algorithms/optimization/mab.hpp: In function ‘std::string oracle::abc_stats_commmands(int, int, int)’:
/home/ratul619/LSoracle/LSOracle/core/algorithms/optimization/mab.hpp:380:1: warning: control reaches end of non-void function [-Wreturn-type]
  380 | }
      | ^
/home/ratul619/LSoracle/LSOracle/core/algorithms/optimization/mab.hpp: In function ‘float oracle::get_results_universe(std::string, int)’:
/home/ratul619/LSoracle/LSOracle/core/algorithms/optimization/mab.hpp:699:1: warning: control reaches end of non-void function [-Wreturn-type]
  699 | }
      | ^
/home/ratul619/LSoracle/LSOracle/core/algorithms/optimization/mab.hpp: In function ‘std::vector<float> oracle::sample(int, int, int, std::vector<std::__cxx11::basic_string<char> >, std::string, std::string, std::vector<std::vector<float> >&, std::vector<std::vector<float> >&, std::vector<std::__cxx11::basic_string<char> >&, int, int, int, int, std::string)’:
/home/ratul619/LSoracle/LSOracle/core/algorithms/optimization/mab.hpp:1173:15: warning: ignoring return value of ‘int system(const char*)’, declared with attribute warn_unused_result [-Wunused-result]
 1173 |         system(commands_vec[i].c_str());
      |         ~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~
/home/ratul619/LSoracle/LSOracle/core/algorithms/optimization/mab.hpp: In function ‘void oracle::bayes_flow_tune(const char*, int, int, int, int, int, int, int)’:
/home/ratul619/LSoracle/LSOracle/core/algorithms/optimization/mab.hpp:1277:52: warning: ignoring return value of ‘int system(const char*)’, declared with attribute warn_unused_result [-Wunused-result]
 1277 |     string remove_log = "rm "+design+".log"; system(remove_log.c_str()); //remove log file if exists
      |                                              ~~~~~~^~~~~~~~~~~~~~~~~~~~
At global scope:
cc1plus: warning: unrecognized command line option ‘-Wno-unneeded-internal-declaration’
make[2]: *** [core/CMakeFiles/lsoracle.dir/build.make:67: core/CMakeFiles/lsoracle.dir/lsoracle.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:4282: core/CMakeFiles/lsoracle.dir/all] Error 2

Any ideas ?? I tried to search , but didnt get any concrete solution

Thanks

yashton, 2022-04-19:
The issue is that our new code has a build flag ENABLE_ABC and ENABLE_STA which isolates the integration for ABC and OpenSTA. If these were set to OFF, parts of the code are turned off that shouldn't. I've fixed this in #81 .

mayank-kabra2001, 2022-04-20:
I am still facing the same issue .. Even after clomning the latest repo. Do you I need to do something else ?

yashton, 2022-04-20:
@mayank-kabra2001 you need to have TCL installed on your machine, this is not a dependency we ship with the tool.

intheRain07, 2022-04-24:
I solve the issue of tcl by:
1. `sudo apt-get install tcl8.6-dev`
2. change `#include <tcl.h>` to `#include <tcl86/tcl.h>`

But another issue comes: 
`Scanning dependencies of target yosys-plugin
[ 96%] Building CXX object yosys-plugin/CMakeFiles/yosys-plugin.dir/oracle.cc.o
/home/rain/LSOracle/yosys-plugin/oracle.cc:54:10: fatal error: kernel/register.h: No such file or directory
   54 | #include "kernel/register.h"
      |          ^~~~~~~~~~~~~~~~~~~
compilation `terminated.`

Any ideas on this problem? 

Thanks.

yashton, 2022-05-02:
If you are trying to compile the yosys plugin, you need to specify the location of yosys using the YOSYS_INCLUDE_DIR cmake variable.

CL-liao, 2022-05-17:
> 你好
> 
> 在发出命令时得到以下错误：
> 
> [ 75%]构建目标 kahypar [ 89%] 构建目标 unit_tests [ 89%] 构建 CXX 对象核心/CMakeFiles/lsoracle.dir/lsoracle.cpp.o 在包含的文件中 /home/ratul619/LSoracle/LSOracle/LSOracle/core/lsoracle.cpp：145： /home/ratul619/LSoracle/LSOracle/core/commands/optimization/optimize.hpp： In member function 'void alice：：optimize_command：：：synth（std：：string）'： /home/ratul619/LSoracle/LSOracle/core/commands/optimization/optimize.hpp：76：27： 错误： 'optimize_resynthesis' 不是 'oracle' 的成员 76 |ntk_result = oracle：：optimize_resynthesismockturtle：：aig_network（partitions_jr， abc_exec）; |^~~~~~~~~~~~~~~~~~~~~~~~~~| >~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~ntk_result = oracle：：optimize_resynthesismockturtle：：aig_network（partitions_jr， abc_exec）; |^ /home/ratul619/LSoracle/LSOracle/core/commands/optimization/optimize.hpp：78：14： error： 'optimization_strategy' 不是 'oracle' 的成员;你是说“optimization_test”吗？ 78 |预言机：：optimization_strategy策略; | |optimization_test /home/ratul619/LSoracle/LSOracle/core/commands/optimization/optimize.hpp：80：3：错误：“策略”未在此范围内声明 80 |策略 = 预言机：：optimization_strategy：:d深度; |^~~~~~~~ /home/ratul619/LSoracle/LSOracle/core/commands/optimization/optimize.hpp：80：22： error： 'oracle：：optimization_strategy' 尚未声明 80 |策略 = 预言机：：optimization_strategy：:d深度; |^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~|~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~策略 = 预言机：：optimization_strategy：：大小; |^~~~~~~~ /home/ratul619/LSoracle/LSOracle/core/commands/optimization/optimize.hpp：82：22： error： 'oracle：：optimization_strategy' 尚未声明 82 |策略 = 预言机：：optimization_strategy：：大小; |^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~|~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~策略 = 预言机：：optimization_strategy：：平衡; |^~~~~~~~ /home/ratul619/LSoracle/LSOracle/core/commands/optimization/optimize.hpp：84：22： error： 'oracle：：optimization_strategy' 尚未声明 84 |策略 = 预言机：：optimization_strategy：：平衡; |^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~| optimize_basic~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~ntk_result = oracle：：optimize_basic（partitions_jr， abc_exec， strategy）; |^~~~~~~~~~~~~~ /home/ratul619/LSoracle/LSOracle/core/commands/optimization/optimize.hpp：87：56： 错误：在“>”标记 87 |之前的预期主表达式ntk_result = oracle：：optimize_basic（partitions_jr， abc_exec， strategy）; |^ /home/ratul619/LSoracle/LSOracle/core/commands/optimization/optimize.hpp：87：83： error： 'strategy' 未在此作用域 87 |ntk_result = oracle：：optimize_basic（partitions_jr， abc_exec， strategy）; |^~~~~~~~
> 
> In file included from /home/ratul619/LSoracle/LSOracle/core/lsoracle.cpp:63: /home/ratul619/LSoracle/LSOracle/core/algorithms/optimization/mab.hpp: In function ‘std::string oracle::abc_stats_commmands(int, int, int)’: /home/ratul619/LSoracle/LSOracle/core/algorithms/optimization/mab.hpp:380:1: warning: control reaches end of non-void function [-Wreturn-type] 380 | } | ^ /home/ratul619/LSoracle/LSOracle/core/algorithms/optimization/mab.hpp: In function ‘float oracle::get_results_universe(std::string, int)’: /home/ratul619/LSoracle/LSOracle/core/algorithms/optimization/mab.hpp:699:1: warning: control reaches end of non-void function [-Wreturn-type] 699 | } | ^ /home/ratul619/LSoracle/LSOracle/core/algorithms/optimization/mab.hpp: In function ‘std::vector oracle::sample(int, int, int, std::vector<std::__cxx11::basic_string >, std::string, std::string, std::vector<std::vector >&, std::vector<std::vector >&, std::vector<std::__cxx11::basic_string >&, int, int, int, int, std::string)’: /home/ratul61 [... cut ...] ion ‘-Wno-unneeded-internal-declaration’ make[2]: *** [core/CMakeFiles/lsoracle.dir/build.make:67: core/CMakeFiles/lsoracle.dir/lsoracle.cpp.o] Error 1 make[1]: *** [CMakeFiles/Makefile2:4282: core/CMakeFiles/lsoracle.dir/all] Error 2
> 
> Any ideas ?? I tried to search , but didnt get any concrete solution
> 
> Thanks

Hello, how did you solve this problem, I re-cloned a copy of the branch of the person who answered this question and tried to compile it, it still failed.What method did you use?

NormandAlexandre, 2022-07-07:
You might want to cmake like this :
cmake -B build . -DCMAKE_BUILD_TYPE=RELEASE -D ENABLE_OPENSTA=ON -DENABLE_ABC=ON
It should work !