lnis-uofu/LSOracle #86: I have a make error。

CL-liao, 2022-06-18:
[image]

NormandAlexandre, 2022-06-24:
Hey, you might want to cmake like this :
cmake -B build . -DCMAKE_BUILD_TYPE=RELEASE -D ENABLE_OPENSTA=ON -DENABLE_ABC=ON 
It should work !

CL-liao, 2022-06-30:
> Hey, you might want to cmake like this : cmake -B build . -DCMAKE_BUILD_TYPE=RELEASE -D ENABLE_OPENSTA=ON -DENABLE_ABC=ON It should work !

Hi,I encountered other problems, and I couldn't find the answer to them by searching online
[image]

CL-liao, 2022-07-09:
> Hello, It looks like possibly you don't have tcl or tcl-dev installed, but have cmake flags set to enable tcl? I would try installing those. If that doesn't resolve it, could you post more complete logs, and tell us about the system you're building it on? I was unable to replicate that error locally or when building on a ubuntu 20.04 docker image w/openSTA dependencies installed. (openroad/ubuntu20-dev:latest)

I am not using ubuntu system, but cenos system.I have tcl installed on my system.
[image]

srtemp, 2022-07-11:
Okay, I'll give it a try on CentOS.  Do you also have the tcl-devel package installed?

CL-liao, 2022-07-12:
> Okay, I'll give it a try on CentOS. Do you also have the tcl-devel package installed?

Thank you, I uninstalled the previous tcl, reinstalled tcl and tcl-devel, and found that the compilation was successful. Thank you for your help.

tianyma, 2022-07-12:
I have this error, any help?
```
[ 94%] Building CXX object core/CMakeFiles/unit_tests.dir/kahypar_config.cpp.o
[ 94%] Building CXX object core/CMakeFiles/unit_tests.dir/kahypar_temp_config.cpp.o
[ 94%] Building CXX object core/CMakeFiles/unit_tests.dir/algorithms/optimization/resynthesis.cpp.o
[ 96%] Building CXX object core/CMakeFiles/unit_tests.dir/utility.cpp.o
[ 96%] Building CXX object core/CMakeFiles/unit_tests.dir/algorithms/partitioning/__tests__/sap_test.cpp.o
[ 96%] Building CXX object core/CMakeFiles/unit_tests.dir/algorithms/partitioning/__tests__/partition_manager_junior_test.cpp.o
[ 96%] Building CXX object core/CMakeFiles/lsoracle.dir/lsoracle.cpp.o
[ 98%] Building CXX object core/CMakeFiles/lsoracle.dir/algorithms/optimization/resynthesis.cpp.o
[ 98%] Building CXX object core/CMakeFiles/lsoracle.dir/kahypar_temp_config.cpp.o
[ 98%] Building CXX object core/CMakeFiles/lsoracle.dir/kahypar_config.cpp.o
[ 98%] Building CXX object core/CMakeFiles/lsoracle.dir/utility.cpp.o
In file included from /lustre/S/matianyun/LSOracle/core/lsoracle.cpp:106:0:
/lustre/S/matianyun/LSOracle/core/commands/output/write_partition.hpp: In member function ‘void alice::write_partition_command::write_part(std::__cxx11::string)’:
/lustre/S/matianyun/LSOracle/core/commands/output/write_partition.hpp:75:21: error: ‘write_child’ is not a member of ‘oracle’
             oracle::write_child<network>(i, partitions, verilog);
                     ^~~~~~~~~~~
/lustre/S/matianyun/LSOracle/core/commands/output/write_partition.hpp:75:40: error: expected primary-expression before ‘>’ token
             oracle::write_child<network>(i, partitions, verilog);
                                        ^
At global scope:
cc1plus: warning: unrecognized command line option ‘-Wno-unneeded-internal-declaration’
make[2]: *** [core/CMakeFiles/lsoracle.dir/lsoracle.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
[ 98%] Linking CXX executable unit_tests
[ 98%] Built target unit_tests
make[1]: *** [core/CMakeFiles/lsoracle.dir/all] Error 2
make: *** [all] Error 2
```

closed 2022-07-27 by NormandAlexandre