lnis-uofu/LSOracle #94: MIG optimization

ratul619, 2022-11-29:
Hi , i generated a blif using yosys for full adder. And when i try to use lsoracle for optimization , doesnt seem to give optimized result. For example , i expect the cout to be just one level - majority between In1 , In2  & Cin. 
-----------------------------------------------
.model fullAdder
.inputs In1 In2 Cin
.outputs Sum Cout
.names $false
.names $true
1
.names $undef
.names In1 In2 $xor$fa.v:25$1_Y
10 1
01 1
.names $xor$fa.v:25$1_Y Cin Sum
10 1
01 1
.names In1 In2 $and$fa.v:27$3_Y
11 1
.names In2 Cin $and$fa.v:27$4_Y
11 1
.names $and$fa.v:27$3_Y $and$fa.v:27$4_Y $or$fa.v:27$5_Y
1- 1
-1 1
.names Cin In1 $and$fa.v:27$6_Y
11 1
.names $or$fa.v:27$5_Y $and$fa.v:27$6_Y Cout
1- 1
-1 1
.end
-----------------------------------------------

Commands used : 

read_blif -m fa.blif
migtune
migscript

MIG logic depth 5 nodes 13
MIG logic depth 5 nodes 13
Final ntk size = 13 and depth = 5
Area Delay Product = 65
Full Optimization: 6ms
Finished optimization

Any more optimization step i am missing ?? Thanks

ratul619, 2023-01-01:
??

yashton, 2023-01-05:
In general, the logic synthesis operations being performed attempt to apply large scale transformations to the logic graph in an attempt to optimize. For something like a single gate or a very small number of gates, these operations don't have a lot of impact and will usually make things significantly worse.